snoplus/orca #266: clicking refresh standard runs produces an error in the debugger

tlatorre-uchicago, 2017-01-23:
If you press the "Refresh Standard Runs" button it produces the following error in the XCode debugger:

```
ERROR: unable to get the receiver data from the DB!
```

I'm not sure what this means as the standard runs seem to be updated, but it's probably worth looking into.

jcaravaca, 2017-01-23:
Does it produce it continuously or just that single time? I tried different combinations locally and I don't seem to be able to reproduce it.

jcaravaca, 2017-03-14:
I haven't seen this recently so I'll consider it fixed.

tlatorre-uchicago, 2017-03-16:
This is still an issue. All you have to do to reproduce is click refresh standard runs while running Orca in XCode.

tlatorre-uchicago, 2017-03-16:
Ah OK. It looks like it happens on the teststand since there is no local couchdb running. When running CouchDB the error doesn't come up every time I click the button, but it does come up once when you first load Orca.

mastbaum, 2018-02-21:
@tlatorre-uchicago are you okay with closing this? Sounds like a valid error about a missing DB, not a bug?

tlatorre-uchicago, 2018-02-21:
Well, it appears even when CouchDB is running when Orca is starting up, so there shouldn't be an error about the DB. I can't even find where in Orca this message is generated, so I'm not really sure what it's reporting.