snoplus/orca #270: If couchdb is down can't start a run at all

marzece, 2017-01-23:
Not even a maintenance or diagnostic run.

marzece, 2017-01-23:
Also a related complaint is that the error message is 
>Error converting JSON response from database: The data couldn’t be read because it isn’t in the correct format.

Which is misleading. This is b/c the relevant lines of code look like 
```obj-c
request = [NSURLRequest requestWithURL:...
    data = [NSURLConnection sendSynchronousRequest:request returningResponse:&response error:&error];
    ret = [[[NSString alloc] initWithData:data ...
    NSDictionary *standardRunTypes = [NSJSONSerialization JSONObjectWithData:[ret dataUsingEncoding:NSUTF8StringEncoding] options:0 error:&error];
    //JSON formatting error
    if (error != nil) {
```

Both `[NSURLConnection.sendSynchronousRequest]` and `[NSJSonSerialization JSONObjectWithData]`
set `error` but it's only checked once.

jcaravaca, 2017-01-23:
Originally there was a 'safe no DB access' run (high threshold run) which was removed from the code. I can work to include a similar thing again for the diagnostic run...
Related with your second comment, it's already corrected in one of my branch which I'm planning to submit as PR.

fbdescamps, 2017-01-27:
Yes, I think we do need a fall-back run in place, diagnostic (maintain current detector settings) sounds good.

tlatorre-uchicago, 2017-03-13:
Fixed in #331.

closed 2017-03-13 by tlatorre-uchicago