snoplus/orca #272: 'Refresh Standard Runs' must be pressed every time ORCA starts.

BenLand100, 2017-01-24:
Is there a reason this doesn't happen automatically? I think it should at least attempt to and fail gracefully if there is an issue with the DB because I can't imagine a scenario where the DB was available that you wouldn't click this button. ( @jcaravaca )

BenLand100, 2017-01-24:
e.g. this fix seems to work fine on the teststand. 
```
diff --git a/Source/Experiments/SNOP/SNOPController.m b/Source/Experiments/SNOP/SNOPController.m
index 18ca838..82ee36c 100755
--- a/Source/Experiments/SNOP/SNOPController.m
+++ b/Source/Experiments/SNOP/SNOPController.m
@@ -240,13 +240,9 @@ snopGreenColor;
     [self refreshRunWordLabels:nil];
     [self runTypeWordChanged:nil];
     //Lock Standard Runs menus at init
-    if([model standardRunType] == nil){
-        [standardRunPopupMenu setEnabled:false];
-        [standardRunVersionPopupMenu setEnabled:false];
-    }
-    else{
-        [self refreshStandardRunsAction:nil];
-    }
+    [standardRunPopupMenu setEnabled:false];
+    [standardRunVersionPopupMenu setEnabled:false];
+    [self refreshStandardRunsAction:nil];
     if(!doggy_icon)
     {
         doggy_icon = [[RunStatusIcon alloc] init];
```

jcaravaca, 2017-01-24:
That's the way it worked in the past, but it was decided that refreshing the SR should be triggered by the user, so that it doesn't complain when opening the experiment in case the CouchDB settings are not correct. Anyways, I don't have a strong opinion on this so feel free to submit the change if it works at the TS.

tlatorre-uchicago, 2017-01-31:
However it was set up before it was failing to refresh very often when ORCA started up and logging lots of errors in the log. Has anything changed since then?

jcaravaca, 2017-01-31:
It should dump an error ('Error querying couchDB, please check the settings are correct and you have connection') if the CouchDB settings are not properly set. I was planning some changes on how errors are handled for the Standard Runs after Eric merges his MTC changes. I can fold this in if we want.

tlatorre-uchicago, 2017-02-21:
This was fixed in #315.

closed 2017-03-02 by tlatorre-uchicago