snoplus/orca #278: HW wizard should disable triggers for low gain PMTs and channels with no PMT attached

tlatorre-uchicago, 2017-01-28:
This is really a database issue, but putting it here as a reminder. The database column which specifies whether a PMT is at HV should include channels with no PMT and low gain PMTs.

tlatorre-uchicago, 2017-01-31:
@fbdescamps how easy is it to add this check to your conversion script?

tlatorre-uchicago, 2017-01-31:
Just realized this needs to be separate from the pulled resistor and pulled cable column since we might want to treat them differently.

fbdescamps, 2017-01-31:
Yes, there is a column now for the low gain tubes in the pmtdb table in detector db. This is currently all zeros but should be updated soon. I'll post a comment here when it is done.

tlatorre-uchicago, 2017-04-14:
This can be solved pretty easily now, but I need to talk with Phil about how we want to implement the HW wizard logic. Should it just use the nominal settings and not enable triggers for channels whose nominal setting is off?