snoplus/orca #284: gt fine delay is never clocked in

tlatorre-uchicago, 2017-01-30:
In SetupGTFineDelay, the fine delay is never clocked in. See https://github.com/snoplus/orca/blob/3b5b83237889fd3ce63ccd4994a998630e84fc25/Source/Objects/Custom%20Hardware/SNO%2B/MTC/ORMTCModel.m#L1569.

tlatorre-uchicago, 2017-01-31:
I originally thought it needed to be clocked in because that is how it's done in the old penn daq code. However, I was testing things out on the teststand today and it seems that writing to the coarse delay and fine delay registers immediately affects the pedestal delay without ever toggling the LOAD_ENPW bit in the control register, so I'm not sure what's going on.

marzece, 2017-02-06:
This certainly isn't an orca issue anymore. #274  removes most MTC register read/writing from orca. It may or may not still be an issue on the mtc server though.

closed 2017-02-06 by marzece