snoplus/orca #290: Limit pulser rate for ECAs

jcaravaca, 2017-02-01:
Since operated by non-experts, we need to check the rate is not dangerous for the DAQ system.

tlatorre-uchicago, 2017-02-01:
Is ECA run while at HV? If so, I guess the one dangerous part would be if you are pedestalling a large number of channels which have triggers enabled. Do you know if SNO left the triggers enabled during an ECA run? If so, it might be worth asking Josh to determine what a safe rate would be to avoid blowing the transistor on the CTC.

jcaravaca, 2017-02-01:
Triggers are never enabled for ECAs. If we are running at high rate and occupancy with no triggers, is there any risk of damaging any DAQ element?

tlatorre-uchicago, 2017-02-01:
Where does the ECA script turn off the channel triggers? If there are no channel triggers then there is no risk, however we would be much less likely to catch a breakdown.

jcaravaca, 2017-02-01:
It is currently coded in the runScript and I just included it in the new ECA code. Eventually we will want them to run intersperse ECAs within a PHYSICS run, but so far the safest option is leave them disabled.
On the other hand, if there is no risk if triggers are OFF we can close this issue for now.

tlatorre-uchicago, 2017-02-02:
That's my understanding. It might be worth limiting the range of the value the user can enter in the text box to something reasonable just to prevent someone accidentally running faster than they wanted.

jcaravaca, 2017-03-14:
Fixed in #255

closed 2017-03-14 by jcaravaca