snoplus/orca #321: Can't create standard runs while in physics runs.

BenLand100, 2017-02-19:
The load standard run, overwrite standard run, threshold fields, and run type word elements are locked out when a physics run is active even in expert mode. This means runs can't be edited or saved while in a physics run. Seems odd to me that we would have to stop a physics run to create a new standard run.

tlatorre-uchicago, 2017-03-13:
Fixed in #331 I think? @jcaravaca

jcaravaca, 2017-03-13:
Not really. You can't change settings while in physics run so having those buttons ON would just create identical copies of the current Physics run. The flow to create a new run is go to maintenance first, change settings and create a new SR.

fbdescamps, 2017-04-04:
I think that flow makes sense? What do you think @BenLand100 ?

BenLand100, 2017-04-04:
It certainly works, but it doesn't solve the underlying "issue" - perhaps it's not really an issue though. The primary annoyance here is that one cannot create standard runs while in physics runs and the root cause is that the run-creation gui is the same as the this-gets-loaded-next-run gui. If we're happy with that then the flow makes sense.

tlatorre-uchicago, 2017-08-11:
I'd like to second this issue. Not sure what the best solution is, but currently you have to actually start a new run in order to save a new standard run. This means you *have* to burn through a run in order to create a new standard run which should be unnecessary.

In my case it came up as a result of wanting to create a new SMELLIE standard run with N100M masked in. It was necessary to first switch to a SMELLIE run, then switch to a diagnostic run (to save the trigger settings), then save the new standard run and then switch into the new standard run. This meant that we essentially wasted two runs just to save a new standard run.

Since saving a new standard run doesn't require actually setting any hardware settings it should be possible to do without going into a maintenance run.

tlatorre-uchicago, 2017-09-25:
Fixed in #440.

closed 2017-09-25 by tlatorre-uchicago