snoplus/orca #324: ORCA should default to locked mode on startup

mnirkko, 2017-02-21:
This UG shift, we restarted ORCA and were able to change expert settings from the start. Since we are 2 newly trained shifters +1 trainee, this should certainly not be possible.

Suggest defaulting to locked mode on startup.

jcaravaca, 2017-02-21:
Was the global security disabled or the panels unlocked? i.e. did you have re-enable security by closing the lock at Preferences->Security or just by closing individual panel locks?

mnirkko, 2017-02-21:
The security was globally disabled, we were able to change trigger 
settings, ramp up crates etc. (under expert guidance via telephone). We 
locked ORCA again using Preferences -> Security.

jcaravaca, 2017-02-21:
Then, this is because an expert saved the .Orca file after disabling the security, which shouldn't have happened.

mnirkko, 2017-02-21:
Well I would still call that a bug if the disabled security can be saved 
in the settings. ;)

jcaravaca, 2017-02-21:
Others would call it a feature :) The problem is that this is part of the core of the ORCA code, used be other experiments. So it might be hard to change. Essentially the global security should *never* be disabled during normal running.

tlatorre-uchicago, 2017-03-13:
Fixed in #331.

closed 2017-03-13 by tlatorre-uchicago