snoplus/orca #327: Ellie Gui seems to crash OCRA

BillyLiggins, 2017-02-22:
While playing with orca, we found that if we open the Data Readout Task window, close it then the  run control window then close it and finally hit the Ellie window, it crashes and we get the following stack trace:

```
Process:         Orca [69041]
Path:            /Users/USER/Desktop/Orca.app/Contents/MacOS/Orca
Identifier:      edu.washington.npl.orca
Version:         11.0h
Code Type:       X86 (Native)
Parent Process:  launchd [170]
Responsible:     Orca [69041]
User ID:         504

Date/Time:       2017-02-22 15:23:30.735 -0500
OS Version:      Mac OS X 10.9.5 (13F1077)
Report Version:  11
Anonymous UUID:  F2920A4D-0B32-784D-5818-F942F88898BE

Crashed Thread:  2  Dispatch queue: com.apple.root.default-priority

Exception Type:  EXC_BAD_ACCESS (SIGBUS)
Exception Codes: KERN_PROTECTION_FAILURE at 0x0000000000000000

VM Regions Near 0:
--> __PAGEZERO             0000000000000000-0000000000001000 [    4K] ---/--- SM=NUL  /Users/USER/Desktop/Orca.app/Contents/MacOS/Orca
    VM_ALLOCATE            0000000000001000-00000000000a6000 [  660K] ---/--- SM=NUL  

Application Specific Information:
objc_msgSend() selector name: release

Thread 0:: Dispatch queue: com.apple.main-thread
0   com.apple.vImage              	0x96334e7e vConvolveCore_Planar8 + 2926
1   com.apple.vImage              	0x96303521 vImageConvolveWithBias_Planar8 + 673
2   com.apple.vImage              	0x9630326f vImageConvolve_Planar8 + 111
3   libRIP.A.dylib                	0x941243e9 symmetric_convolve + 615
4   libRIP.A.dylib                	0x94123fc6 RIPLayerSymmetricConvolve + 692
5   libRIP.A.dylib                	0x94123c7a RIPLayerGaussianBlur + 769
6   libRIP.A.dylib                	0x9412887c rips_s_BltImage + 568
7   libRIP.A.dylib                	0x94119a59 ripc_RenderImage + 199
8   libRIP.A.dylib                	0x94117dbb ripc_DrawImage + 1051
9   com.apple.CoreGraphics        	0x952c43fc CGContextDrawImage + 429
10  com.apple.AppKit              	0x97362c35 __74-[NSImageRep drawInRect:fromRect:operation:fraction:respectFlipped:hints:]_block_invoke + 920
11  com.apple.AppKit              	0x97362886 NSUsingGraphicsStateForHints_drawWithBlock_ + 236
12  com.apple.AppKit              	0x97362648 -[NSImageRep drawInRect:fromRect:operation:fraction:respectFlipped:hints:] + 895
13  com.apple.AppKit              	0x97362075 __71-[NSImage drawInRect:fromRect:operation:fraction:respectFlipped:hints:]_block_invoke + 1102
14  com.apple.AppKit              	0x972eae78 -[NSImage _usingBestRepresentationForRect:context:hints:body:] + 167
15  com.apple.AppKit              	0x97361b88 -[NSImage drawInRect:fromRect:operation:fraction:respectFlipped:hints:] + 1467
16  com.apple.AppKit              	0x97377009 -[NSImage drawInRect:fromRect:operation:fraction:] + 111
17  com.apple.AppKit              	0x97376f8f -[NSImage drawAtPoint:fromRect:operation:fraction:] + 193
18  edu.washington.npl.orca       	0x000cceac 0xa6000 + 159404
19  edu.washington.npl.orca       	0x000ccb3f 0xa6000 + 158527
20  edu.washington.npl.orca       	0x000ccaf6 0xa6000 + 158454
21  edu.washington.npl.orca       	0x000c3d08 0xa6000 + 122120
22  edu.washington.npl.orca       	0x000c4bfd 0xa6000 + 125949
23  edu.washington.npl.orca       	0x000c4b20 0xa6000 + 125728
24  com.apple.AppKit              	0x97351298 -[NSView _drawRect:clip:] + 3816
25  com.apple.AppKit              	0x9734fccf -[NSView _recursiveDisplayAllDirtyWithLockFocus:visRect:] + 1829
26  com.apple.AppKit              	0x9734fffc -[NSView _recursiveDisplayAllDirtyWithLockFocus:visRect:] + 2642
27  com.apple.AppKit              	0x9734fffc -[NSView _recursiveDisplayAllDirtyWithLockFocus:visRect:] + 2642
28  com.apple.AppKit              	0x9734fffc -[NSView _recursiveDisplayAllDirtyWithLockFocus:visRect:] + 2642
29  com.apple.AppKit              	0x9734fffc -[NSView _recursiveDisplayAllDirtyWithLockFocus:visRect:] + 2642
30  com.apple.AppKit              	0x9734e278 -[NSView _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 1264
31  com.apple.AppKit              	0x9734f0b6 -[NSView _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 4910
32  com.apple.AppKit              	0x9734f0b6 -[NSView _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 4910
33  com.apple.AppKit              	0x9734d8f0 -[NSThemeFrame _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 290
34  com.apple.AppKit              	0x9734ac57 -[NSView _displayRectIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:] + 3645
35  com.apple.AppKit              	0x9732872c -[NSView displayIfNeeded] + 1414
36  com.apple.AppKit              	0x9734674c -[NSWindow displayIfNeeded] + 324
37  com.apple.AppKit              	0x97390365 _handleWindowNeedsDisplayOrLayoutOrUpdateConstraints + 1107
38  com.apple.AppKit              	0x97a0015b __83-[NSWindow _postWindowNeedsDisplayOrLayoutOrUpdateConstraintsUnlessPostingDisabled]_block_invoke1381 + 58
39  com.apple.CoreFoundation      	0x9071086d _runLoopObserverWithBlockContext + 29
[... 782 lines cut ...]
===========                      =======
TOTAL                               1.3G

Model: MacPro6,1, BootROM MP61.0116.B05, 4 processors, Quad-Core Intel Xeon E5, 3.7 GHz, 12 GB, SMC 2.20f18
Graphics: AMD FirePro D300, AMD FirePro D300, PCIe, 2048 MB
Graphics: AMD FirePro D300, AMD FirePro D300, PCIe, 2048 MB
Memory Module: DIMM1, 4 GB, DDR3 ECC, 1867 MHz, 0x80AD, 0x484D54343531553741465238432D52442020
Memory Module: DIMM2, 4 GB, DDR3 ECC, 1867 MHz, 0x80AD, 0x484D54343531553741465238432D52442020
Memory Module: DIMM3, 4 GB, DDR3 ECC, 1867 MHz, 0x80AD, 0x484D54343531553741465238432D52442020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x135), Broadcom BCM43xx 1.0 (6.30.223.154.65)
Bluetooth: Version 4.2.7f4 15112, 3 services, 15 devices, 1 incoming serial ports
Network Service: Ethernet 1, Ethernet, en0
PCI Card: AMD FirePro D300, Display Controller, Slot-2
PCI Card: AMD FirePro D300, Display Controller, Slot-1
Serial ATA Device: APPLE SSD SM0256F, 251 GB
USB Device: Hub
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: Keyboard Hub
USB Device: USB Optical Mouse
USB Device: Apple Keyboard
Thunderbolt Bus: Mac Pro, Apple Inc., 19.2
Thunderbolt Bus: Mac Pro, Apple Inc., 19.2
Thunderbolt Bus: Mac Pro, Apple Inc., 19.2
```

EdLeming, 2017-02-22:
hey @BillyLiggins, good catch! It's quite a specific click sequence you  detail, have you managed to re-produce?

BillyLiggins, 2017-02-23:
You're right it turns out the sequence is not needed, if we click on the Ellie window orca crashes. We have reproduced it ~5 times. The trace on the direct Ellie click is below, I don't know if they are the same. 

```Process:         Orca [70901]
Path:            /Users/USER/Desktop/Orca.app/Contents/MacOS/Orca
Identifier:      edu.washington.npl.orca
Version:         11.0h
Code Type:       X86 (Native)
Parent Process:  launchd [170]
Responsible:     Orca [70901]
User ID:         504

Date/Time:       2017-02-22 19:35:41.218 -0500
OS Version:      Mac OS X 10.9.5 (13F1077)
Report Version:  11
Anonymous UUID:  F2920A4D-0B32-784D-5818-F942F88898BE

Crashed Thread:  7  Dispatch queue: com.apple.root.default-priority

Exception Type:  EXC_BAD_ACCESS (SIGSEGV)
Exception Codes: KERN_INVALID_ADDRESS at 0x00000000f0000020

VM Regions Near 0xf0000020:
    CG shared images       00000000c871f000-00000000c8727000 [   32K] r--/r-- SM=SHM  
--> 
    Submap                 00000000ffff0000-00000000ffff1000 [    4K] r--/r-- SM=PRV  process-only VM submap

Application Specific Information:
objc_msgSend() selector name: release

Thread 0:: Dispatch queue: com.apple.main-thread
0   com.apple.CoreFoundation      	0x906a4081 CFRelease + 961
1   com.apple.CoreGraphics        	0x952a43af CGSReleaseRegion + 17
2   com.apple.AppKit              	0x97380f4b -[NSRegion subtractRegion:] + 212
3   com.apple.AppKit              	0x97383e5d -[NSRegion subtractRect:] + 130
4   com.apple.AppKit              	0x979f004e NSViewSetNeedsDisplayInRect + 728
5   com.apple.AppKit              	0x971fa4b5 -[NSView setNeedsDisplayInRect:] + 35
6   com.apple.WebKit              	0x93dc9c0f setNeedsDisplayInRect(NSView*, objc_selector*, _NSRect) + 223
7   com.apple.AppKit              	0x973242dd -[NSTextView setNeedsDisplayInRect:avoidAdditionalLayout:] + 875
8   com.apple.AppKit              	0x972bc5b1 -[NSLayoutManager _primitiveInvalidateDisplayForGlyphRange:] + 4024
9   com.apple.AppKit              	0x972bb0d9 -[NSLayoutManager invalidateDisplayForCharacterRange:] + 433
10  com.apple.AppKit              	0x972ccb8a -[NSLayoutManager(NSPrivate) _invalidateDisplayIfNeeded] + 70
11  com.apple.AppKit              	0x972c4393 -[NSLayoutManager textContainerForGlyphAtIndex:effectiveRange:] + 378
12  com.apple.AppKit              	0x9745ae39 -[NSTextView(NSPrivate) _scrollRangeToVisible:forceCenter:] + 456
13  com.apple.AppKit              	0x9745ac67 -[NSTextView scrollRangeToVisible:] + 55
14  edu.washington.npl.orca       	0x00027fee 0x1a000 + 57326
15  libobjc.A.dylib               	0x990572af -[NSObject performSelector:withObject:] + 70
16  com.apple.Foundation          	0x92c35712 __NSThreadPerformPerform + 318
17  com.apple.CoreFoundation      	0x90710b5f __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 15
18  com.apple.CoreFoundation      	0x9070195b __CFRunLoopDoSources0 + 235
19  com.apple.CoreFoundation      	0x9070105e __CFRunLoopRun + 1022
20  com.apple.CoreFoundation      	0x907009ea CFRunLoopRunSpecific + 394
21  com.apple.CoreFoundation      	0x9070084b CFRunLoopRunInMode + 123
22  com.apple.HIToolbox           	0x9a461b5d RunCurrentEventLoopInMode + 259
23  com.apple.HIToolbox           	0x9a4618e2 ReceiveNextEventCommon + 526
24  com.apple.HIToolbox           	0x9a4616bd _BlockUntilNextEventMatchingListInModeWithFilter + 92
25  com.apple.AppKit              	0x971ea349 _DPSNextEvent + 1602
26  com.apple.AppKit              	0x971e9870 -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 119
27  com.apple.AppKit              	0x971dc15c -[NSApplication run] + 727
28  com.apple.AppKit              	0x971c4ff8 NSApplicationMain + 1165
29  libdyld.dylib                 	0x9bba6701 start + 1

Thread 1:: Dispatch queue: com.apple.root.default-priority
0   libsystem_kernel.dylib        	0x9693bfb6 semaphore_wait_trap + 10
1   libdispatch.dylib             	0x9c26b0a4 _dispatch_semaphore_wait_slow + 274
2   libdispatch.dylib             	0x9c26af1f dispatch_semaphore_wait + 37
3   com.apple.CFNetwork           	0x914844a3 CFURLConnectionSendSynchronousRequest + 414
4   com.apple.Foundation          	0x92d91675 +[NSURLConnection sendSynchronousRequest:returningResponse:error:] + 184
5   edu.washington.npl.orca       	0x00816279 0x1a000 + 8372857
6   edu.washington.npl.orca       	0x0080d29c 0x1a000 + 8336028
7   libdispatch.dylib             	0x9c26976b _dispatch_call_block_and_release + 15
[... 817 lines cut ...]
TOTAL, minus reserved VM space      1.3G
 

Model: MacPro6,1, BootROM MP61.0116.B05, 4 processors, Quad-Core Intel Xeon E5, 3.7 GHz, 12 GB, SMC 2.20f18
Graphics: AMD FirePro D300, AMD FirePro D300, PCIe, 2048 MB
Graphics: AMD FirePro D300, AMD FirePro D300, PCIe, 2048 MB
Memory Module: DIMM1, 4 GB, DDR3 ECC, 1867 MHz, 0x80AD, 0x484D54343531553741465238432D52442020
Memory Module: DIMM2, 4 GB, DDR3 ECC, 1867 MHz, 0x80AD, 0x484D54343531553741465238432D52442020
Memory Module: DIMM3, 4 GB, DDR3 ECC, 1867 MHz, 0x80AD, 0x484D54343531553741465238432D52442020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x135), Broadcom BCM43xx 1.0 (6.30.223.154.65)
Bluetooth: Version 4.2.7f4 15112, 3 services, 15 devices, 1 incoming serial ports
Network Service: Ethernet 1, Ethernet, en0
PCI Card: AMD FirePro D300, Display Controller, Slot-2
PCI Card: AMD FirePro D300, Display Controller, Slot-1
Serial ATA Device: APPLE SSD SM0256F, 251 GB
USB Device: Hub
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: Keyboard Hub
USB Device: USB Optical Mouse
USB Device: Apple Keyboard
Thunderbolt Bus: Mac Pro, Apple Inc., 19.2
Thunderbolt Bus: Mac Pro, Apple Inc., 19.2
Thunderbolt Bus: Mac Pro, Apple Inc., 19.2
```

tlatorre-uchicago, 2017-03-13:
Fixed in #328.

closed 2017-03-13 by tlatorre-uchicago