snoplus/orca #330: Hardware wizard totally violates principle of least surprise.

BenLand100, 2017-02-25:
Tried to set manual corrections to channel thresholds for crate 16 to zero... ended up getting small positive and negative numbers instead, and as a bonus the hardware state of the triggers returned to that of the last run with no warning. No idea how that even happens.

boardhead, 2017-02-27:
For the hardware wizard to function properly ORCA must first agree with the current detector state.  The best way I could think of for doing this was to load the complete detector state from the detector database before the wizard operation.  The result is that ORCA is forced to agree with the detector state when using the wizard (not the other way around, which was the previous behaviour).  I agree that would could better inform the operator about discrepancies, but it is unclear how it would be best to implement this.

BenLand100, 2017-02-27:
Are you sure it's the current detector state as opposed the state at the start of the last run that ORCA must agree with? Because it seemed to work more as expected if I restarted the run before doing HW wizard actions. Otherwise if I made any trigger changes in ORCA (and pressed load hardware, etc) without restarting the run they would get overwritten after using the HW wizard.

BenLand100, 2017-02-28:
Following up on this: Tony and I realized today that what was going on with the channel triggers is that crate triggers were turned off for some crates crates and any hardware wizard interaction set the current state for the n20 and n100 in the fec window to disabled if the crate triggers were off. That's probably a bug but maybe not a hardware wizard issue. 

The problem with setting thresholds to zero resulting in random small (positive/negative) manual corrections persists.

tlatorre-uchicago, 2017-03-02:
I think one thing that would be nice is if the hardware wizard only operated on the crates that are set. Currently if you perform a hardware wizard action on a single crate, it reloads the settings for all the crates.