snoplus/orca #335: can't close Orca

tlatorre-uchicago, 2017-03-04:
Currently, it's not possible to quit Orca during a run by going to Orca -> Quit Orca. I suspect this is due to exceptions being thrown when objects are deallocated, but I'm not 100% sure. This needs to be fixed because it is part of the handover procedure.

tlatorre-uchicago, 2017-03-05:
Here are a couple of segfaults I have seen on the teststand when trying to quit while a run is going on:

[image]
[image]

tlatorre-uchicago, 2017-03-05:
I was able to get Orca to quit while a run was going by calling `[runControl setRunningState:eRunStopped]` when a `OROrcaAboutToQuitNotice` notification is posted.

fbdescamps, 2017-04-04:
@boardhead do you have an update on this issue? Thanks!

tlatorre-uchicago, 2017-04-14:
Fixed in https://github.com/snoplus/orca/commit/49ff707c4caf73462da22465f8e9bf1bca70501f.

closed 2017-04-14 by tlatorre-uchicago