snoplus/orca #352: simplify run start, stop, resync

tlatorre-uchicago, 2017-03-13:
It was brought up at the collaboration meeting that there is a lot of confusion over the start, stop, restart, and resync buttons.

One of the only reasons for the restart button is that there is a little less dead time between runs since we don't have to wait for the buffers to flush.

One solution is that we only have start and stop buttons, and both of these do hard stops and starts, and the only time the buffers are not stopped are during a continuous run rollover which should be most of the time during stable running.

jcaravaca, 2017-03-13:
This requires GUI changes and I was waiting for Ed to implement his ELLIE modifications. Other than that should be easy to implement.

EdLeming, 2017-03-14:
I'll get those in a PR asap.

tlatorre-uchicago, 2017-04-14:
See the discussion in #367. It was decided that we would not implement this.

closed 2017-04-14 by tlatorre-uchicago