snoplus/orca #356: Bug in displayed Run Version (SNO+ Control Panel)

vlozza, 2017-03-14:
Reading from the SNO+ Control Panel:
Run type is PHYSICS.
If the run is started with a Run Version different from DEFAULT this is initially displayed in Run Version. However, when the run rolls over (new run starts) the run Version is automatically set to "DEFAULT" even if the detector state doesn't change (read from the monitoring/state/0 and the MTC card).

Example: tonight we were running in LOWER_NHITS_PHYSICS_3 (nHItt100Med set to 13). This was initially displayed in the Run Version (Run Version: LOWER_NHITS_PHYSICS_3). However, when a new run automatically started the displayed Run Version changed to DEFAULT (Run Version: DEFAULT). However, a careful look at the Nhit100Med value, at the builder rate, and so on shows the setting as before.

jcaravaca, 2017-03-15:
I've seen the same when starting an ECA TSLP run: the TSLP version settings were loaded but the SNO+ panel claimed it was running an ECA DEFAULT run. I'll look into fixing this.

tlatorre-uchicago, 2017-08-16:
Suspect that this was due to the Orca file being saved on the desktop. Someone should reopen this if it happens again.

closed 2017-08-16 by tlatorre-uchicago