snoplus/orca #363: ORCA crashes when run type is selected

terezakroupa, 2017-03-18:
Observed an ORCA crash twice while attempting to switch to physics run after ramping up. See shift report [here](http://snopl.us/shift/view/9d882118c5b64f60961313f27d1eee59)

@jlidgard and @jcaravaca also saw this day before yesterday during LV SMELLIE running.

See log in attachment.
[orca_crash_1.txt](https://github.com/snoplus/orca/files/852065/orca_crash_1.txt)

marzece, 2017-03-19:
We saw this crash again. Just before it happened Jeff had clicked the run type selectrion drop down and a dialog popped up asking if we wanted to save a new run with some weird name. See attached screen shot below. Jeff clicked cancel but Orca promptly crashed after that.
[image]

jlidgard, 2017-03-19:
We've seen a couple of crashes this evening. The one (and another, it happened again) above was interesting. When I clicked the 'Run Name' dropdown arrow, a string appeared in the box '53.6MA' and ORCA immediately popped a dialog saying 'do you want to save a new run type' (or similar). When I pressed cancel on this box, it crashed. 
Pressing the button is accessing a rogue string/bit of memory?

fbdescamps, 2017-04-04:
@jcaravaca : I believe that this one has been resolved? If so, please close this issue. Thanks!