snoplus/orca #369: caen and tubii data need to be in standard run

tlatorre-uchicago, 2017-03-20:
The CAEN and TUBii settings should be a part of the standard run. There are several reasons for this:

- for calibration runs, the CAEN window length needs to change and we may want to enable extra channels in the readout
- the gain for the CAEN channels is set in TUBii and also might need to change for certain calibration runs

We could try to pick a minimal subset of the settings which are needed, but since there are only 10 CAEN settings and ~16 tubii settings, we should probably just add them all.

In addition, we will need to add an `initAtRunStart` method to TUBii which loads all of the GUI settings -> hardware.

fbdescamps, 2017-04-04:
@jcaravaca ?

tlatorre-uchicago, 2017-08-16:
Fixed in #412.

closed 2017-08-16 by tlatorre-uchicago