snoplus/orca #375: Standard run selector didn't select "Default" by default

marzece, 2017-03-31:
I was switcching from a physics non-default run to a maintenance run and it set the version to THRES_SCAN instead of default.

jcaravaca, 2017-04-01:
You were in expert mode, right? In that case, it selects the first of the list. Why would you find useful to switch automatically to default?

fbdescamps, 2017-04-04:
I think, if it easy to do, it would make sense to select 'default' by default when selecting standard runs?

marzece, 2017-04-04:
I was in expert mode. I think it would useful b/c some of our less diligent shifters might just quickly switch to maintenance and not notice the version is "LOW_NHIT_TESTING_WILL_KILL_EVERYONE_V2" or something like that.

tlatorre-uchicago, 2017-08-16:
@jcaravaca has this been changed?

jcaravaca, 2017-08-17:
Mostly. It currently selects default if the previous selected version is not found, which is true as long as we don't go from typeA/versionA to typeB/versionA, and we don't have versions named the same for different SRs atm.

tlatorre-uchicago, 2017-08-17:
I think that's reasonable.

closed 2017-08-17 by tlatorre-uchicago