snoplus/orca #382: user shouldn't be able to change trigger bits for standard run

tlatorre-uchicago, 2017-04-07:
It's happened quite a few times now that the physics standard run ends up with the maintenance bit checked, because the normal procedure is to switch to maintenance run, adjust thresholds, and then save the new standard run. I think these need to be hardcoded for each standard run to avoid this.

tlatorre-uchicago, 2017-04-24:
Just a note that this happened again and we've been running for a long time with "PHYSICS" selected but the run type word is marked as maintenance.

mastbaum, 2018-03-30:
Hi @jcaravaca, have you had a chance to look into this at all?

jcaravaca, 2018-03-30:
I think this is a tricky one since if we don't allow to tweak the run type word in ORCA, then we can't change it at all (unless one hacks into CouchDB). That means we won't be able to define new standard runs, only new standard run versions. I could prevent the user/expert to create new standard runs at all, and reserve that task to only me (which I could do offline just pushing tables to CouchDB). In this way we would be able to create only versions, which will show the same run type. I don't think we ever reached a consensus on this, so we could rediscuss this on the next DAQ call.