snoplus/orca #385: SNOP GUI's global "Triggers Off" might have turned cmos rate polling off

marzece, 2017-04-11:
I had had cmos rate polling on for every crate and I noticed that at some point that had gotten turned off. The only operation I'd done recently was turn triggers off for the entire detector so I suspect that's what caused the issue. I didn't test it though so I can't be sure.

tlatorre-uchicago, 2017-05-29:
It does indeed do that. See here: https://github.com/snoplus/orca/blob/99bc491c1a7467c00ef4b5d00bdb3515c20ecfaf/Source/Experiments/SNOP/SNOPModel.m#L1761. No idea why this was implemented like this.

jcaravaca, 2017-05-31:
Panic down turns polling OFF in the first place (as in the XL3 model) and the triggers OFF button inherited the same behavior. Probably not needed.

tlatorre-uchicago, 2017-06-27:
Fixed in #406.

closed 2017-06-27 by tlatorre-uchicago