snoplus/orca #393: consider removing or locking out global triggers off button

tlatorre-uchicago, 2017-04-24:
There have been a few times now where operators have hit the global triggers off button in response to a detector problem. Since there are very few circumstances where it's ever a good idea to turn all of the channel triggers off and in those cases an expert can turn them off for each crate by hand (now with the hotkeys this is fairly simple), we should consider getting rid of this button.

tannerbk, 2017-05-14:
I think this button has some use still for standard operators. For example during a big light leak (like we just had) it would probably have been a good idea for the operators to do a global triggers off.

tlatorre-uchicago, 2017-05-15:
Wouldn't it have been better to panic down in that situation? I guess it depends on how often there will be a situation where there is enough light in the detector to cause issues with the trigger system vs the actual PMTs.

One of the worries is that in a light leak or breakdown scenario, the operator hits the global triggers off button and the detector quiets down (i.e. the event rate goes down) and then the operator thinks things are in a safe state but in fact the light leak has not been fixed.

It may be the case that any light leak big enough to damage the PMTs will cause the ESUM triggers to fire in which case this isn't as big of a concern.

tlatorre-uchicago, 2017-06-28:
This doesn't seem to be a problem much any more. I think the shifters have figured out how to do things now, and this button can definitely be handy when at LV, so I'm going to close.

closed 2017-06-28 by tlatorre-uchicago