snoplus/orca #408: Owl triggers say they are on when HV is off

BillyLiggins, 2017-05-24:
Today i ramped down power supply 16b and panicked a bit when I saw that the triggers were still on for that supply when at LV in the HV tab.  It wasn't until Erica told me that this was not the case that I didn't think I had blown a CTC. Thank you.

jcaravaca, 2017-05-24:
Is that in the HV master?

BillyLiggins, 2017-05-24:
No, in the xl3 HV tab.

BillyLiggins, 2017-05-24:
In fact it may have been in the HV master but i only saw it in the xl3 HV tab

ecaden, 2017-05-24:
But it also shows triggers "ON" in the HV master, which doesn't seem to be related to the HV status for that 16B.

jcaravaca, 2017-05-24:
So to summarize: HV master and XL3 HV tabs show 'Triggers ON' while the trigger interlock for that crate and the HV were actually OFF. Is that correct?

BenLand100, 2017-05-25:
This is a historical thing that just hasn't been changed (yet). The 16B HV info pane on the XL3 window shows "Triggers: ON" if crate 16's triggers are on. In reality that indicator has nothing to do with 16B, since its channels are on slot 15 of crates 3, 13, and 18. We haven't been worried about ensuring those triggers are off when ramping 16B because it's "only one card in each crate" which I've been told is not an issue due to the relatively fewer number of channels that might go crazy per ctc. 

That being said, the erroneous label should be removed the next time that gui is tweaked.