snoplus/orca #420: Incorrect run state reported by ORCA after handover

rcl11, 2017-06-07:
After handover from UG->AG. Detector was in transition run at end of calibration work. When opening ORCA above ground it reported the trigger settings of the transition run but thought we were in a physics run (and DB trigger values for physics run were reported). Screenshot below. Day shifters noticed similar but not exactly same issue when then did handover from AG->UG, see #419 

[image]

tlatorre-uchicago, 2017-06-12:
Are you sure the u/g shifters hadn't selected physics run before saving Orca and doing the handover? The run type selected in the drop down menu and the trigger values displayed in the DB column are always those of what was last *selected* and not what was actually running.

rcl11, 2017-06-13:
Well from memory it was @vfisher and @mmdepatie underground so maybe they could comment. It's entirely possible that "physics" was selected on the drop down list underground when they saved given that we were about to change into a physics run following the calibration work. However I think what surprised us more was the fact that it says "Run type: physics" in the "run status" menu, rather than the drop down list.

tlatorre-uchicago, 2017-06-27:
This was most likely due to someone having saved the Orca file on the Desktop on surface.

closed 2017-06-27 by tlatorre-uchicago