snoplus/orca #421: Put LO into standard run

tannerbk, 2017-06-12:
I understand LO is a global setting and not something that changes between run types, but it might be worth putting the lockout length into the standard physics run so that it doesn't accidentally get changed, as it did recently. Maybe there is a better way to ensure the LO is 420 during physics that I'm unaware of.

tlatorre-uchicago, 2017-06-27:
As a temporary measure, I've hardcoded it so that the lockout is set to 420 ns at the start of every run. See this commit: https://github.com/snoplus/orca/commit/ac38fd7690dd23687fcc139d91a8108f4f5b5146. When this is implemented, we should delete that line.

jcaravaca, 2017-08-08:
Fixed in #412. Will close when merged.

closed 2017-08-16 by tlatorre-uchicago