snoplus/orca #429: tubii ip address isn't loaded from file

tlatorre-uchicago, 2017-06-27:
On the teststand, if you add TUBii to the experiment it comes up with the default ip address of the tubii underground (this is dangerous because it means when you change settings on the teststand, they get changed underground).

I changed the hostname to tubii-teststand.sp.snolab.ca and saved the document, but when reopening it, it comes back with the hardcoded ip address of tubii underground.

tlatorre-uchicago, 2017-07-27:
@icoulter I think this is going to be important once TUBii is included in the standard runs as at that time the TUBii settings will get loaded on every run start, and so right now running that on the teststand will overwrite all of the TUBii settings underground.

icoulter, 2017-08-08:
Thanks. This seems to work.
I think we need to make sure experts are aware of the correct IP and Port in case this ever needs set by hand (which I presume is an expert operation).

closed 2017-08-15 by tlatorre-uchicago