snoplus/orca #430: tubii pgt mask error in debugger

tlatorre-uchicago, 2017-06-27:
When compiling Orca in XCode on operag1 which is running OSX 10.10.5, I got an error in the debugger about not being able to connect to the tubii pgt mask button. I'm unable to reproduce on the teststand. I'll try and grab a screenshot next time.

@icoulter, if you happen to have a mac running 10.10.5, could you try and reproduce this?

icoulter, 2017-06-27:
My mac does happen to be running 10.10.5. I don't find any issue though.
A screenshot would help.

tlatorre-uchicago, 2017-06-28:
[image]

tlatorre-uchicago, 2017-06-28:
This happened only after I clicked on the TUBii icon.

icoulter, 2017-06-28:
Yeah. I see it now. Will try to fix it.

icoulter, 2017-06-28:
See #431 
I'd linked the second button to the controller, changing the name of the first in the process from button->start and hadn't tidied up after myself.

tlatorre-uchicago, 2017-06-29:
Fixed in #431.

closed 2017-06-29 by tlatorre-uchicago