snoplus/orca #439: tubii ping doesn't acknowledge that the ping was received

tlatorre-uchicago, 2017-08-15:
In the TUBii GUI there is a "Ping" button, but when you press it it just says "Sending ping to TUBii" in the status window, but doesn't tell you whether it successfully connected or not.

icoulter, 2017-08-15:
I thought it wrote on the DAQ log. Also, I'm pretty sure it issues an error if there isn't a TUBii connected to that address.

tlatorre-uchicago, 2017-08-15:
It would be nice to just print "ping successful" or something. "Ping sent" sounds like it sent it, but never received a response.

icoulter, 2017-08-15:
Fair enough. I'll add it to the list and slip it in with the next firmware update.

tlatorre-uchicago, 2017-08-16:
I mean in the Orca dialog window. For example, on the teststand, I was trying to see if I successfully connected to the fake tubii server and this is what the status window shows when you click ping:

[image]

I think just after "Sending ping to TUBii" it should print if it successfully got a response back.

icoulter, 2017-08-16:
Yes. I'm agreeing with you. But the problem is that the firmware/orca aren't coordinating any response so it'll take a firmware update as well. I'll change it next time I need to do something else as its a pretty minor issue.

tlatorre-uchicago, 2017-09-15:
Fixed in #445.

closed 2017-09-15 by tlatorre-uchicago