snoplus/orca #442: tubii should match hardware at the start of a run

tlatorre-uchicago, 2017-08-16:
Currently the TUBii GUI comes up with default values in the model, and so when you start up Orca it looks like this:

[image]

And so it looks like the current values are different than the expected values.

Either the TUBii model should save all the GUI settings to the Orca file and reload them or match with hardware when starting up (ideally both, first the orca settings and then match hardware).

estherturner, 2018-05-04:
Just wanted to let people know that this issue is occasionally causing some confusion for shifters and resulting in extra restarts being done so that the values match.

tlatorre-uchicago, 2019-01-17:
Just to comment, the values not matching should not be fixed by restarting the run. This is a purely GUI bug and doesn't reflect the fact that the hardware settings are actually wrong.