snoplus/orca #443: tellie does not stop firing when run changes

tlatorre-uchicago, 2017-08-17:
Today we were running TELLIE and forgot to stop firing the fibre when we switched into a physics run. The fibre continued to fire during the physics run.

The TELLIE code needs to post a run stop change wait notification before the run stops and stop firing before the run changes.

EdLeming, 2017-08-30:
Oh, I didn't see this. Thanks for tagging me. @tlatorre-uchicago, could you confirm the sequence of button clicks on this occation? 

The stopRunAction method of SNOPController.m does have a check for this already. That said, now I think about it there is also the situation where one could just change the standard run selection and hit the start / restart button. In that case the check I put in place will not be called.

tlatorre-uchicago, 2017-08-30:
I don't remember for sure, but we probably did click start run and not stop run.

ecaden, 2017-08-31:
I noticed this happening again when trying to recover from a breakdown/LV issue on crate 11 this morning.

EdLeming, 2017-08-31:
Yeah, If a user restarts / resyncs into a maintenance run with out explicitly hitting the stop *ellie button this will happen atm. I'll put up a PR later today.

tlatorre-uchicago, 2017-10-31:
Fixed in #446.

closed 2017-10-31 by tlatorre-uchicago