snoplus/orca #448: ellie model not releasing run state change wait

tlatorre-uchicago, 2017-09-08:
We were running a SMELLIE run and the server timed out when switching intensities:

```
17-09-08 13:45:54 Run 104807 started(cold start).
17-09-08 13:45:54 DataTaking Thread Started
17-09-08 13:46:09 [SMELLIE] config file (version 22) sucessfully loaded
17-09-08 13:46:24 [SMELLIE]:Setting up a SMELLIE Run
17-09-08 13:46:24 Sending GetAsyncTriggerMask to TUBii
17-09-08 13:46:24 [SMELLIE]: Running in MASTER mode
17-09-08 13:46:47 [SMELLIE]: Fire sequence requested for laser: PQ440
17-09-08 13:46:47 [SMELLIE]: Fire sequence requested for fibre: FS093
17-09-08 13:46:47 Sending SetSmellieDelay 500 to TUBii
17-09-08 13:52:47 [SMELLIE]: Problem with smellie server request: The request timed out.
17-09-08 13:52:47 [SMELLIE]: Sent to err statement. Stopping fire sequence.
17-09-08 13:52:47 CAEN settings are different in this standard run. Resyncing...
17-09-08 13:52:47 Standard run MAINTENANCE (DEFAULT) settings loaded.
17-09-08 13:54:56 DataTaking Thread Exited
17-09-08 13:54:56 Transfer queue cleared successfully
17-09-08 13:54:56 Waiting for transfer thread to exit
17-09-08 13:54:56 Transfer thread exited
17-09-08 13:54:56 Closing dataFile: ~/Data/openFiles/Run104807
17-09-08 13:54:56 There is more than one file in ~/Data/openFiles. Will attempt to move all of them.
17-09-08 13:54:56 Move ~/Data/openFiles/Run-1 --> ~/Data/Run-1  <Failed>
17-09-08 13:54:56 Move ~/Data/openFiles/Run104807 --> ~/Data/Run104807  <OK>
17-09-08 13:54:56 1 file failed to copy out of the openFiles folder. You will have to move it manually
17-09-08 13:54:56 Final end of run cleanup
17-09-08 13:54:56 Run 104807 stopped.
17-09-08 13:54:56 ------------------Error Summary---------------------
17-09-08 13:54:56 Errors   count: 0
17-09-08 13:54:56 ----------------------------------------------------
```

However, the run was never able to actually stop because of an ELLIE notification to not change the run. We waited 3 minutes and still nothing.

I had to manually force clear the notifications from the Orca run control window to get things to move on.

After the new run started we got this message in the log:

```
17-09-08 14:14:47 [SMELLIE]: Stopped sending keep-alive to interlock server
17-09-08 14:14:53 [SMELLIE]: Smellie laser interlock server disarmed
17-09-08 14:14:53 Sending SetSmelliePulser 1 0.1 0 to TUBii
17-09-08 14:14:53 [SMELLIE]: Run sequence stopped.
```

At this time the average nhit dropped from 24 -> 7 and so it seems like light was still being fired into the detector.

EdLeming, 2017-09-08:
This is a problem is due to SMELLIE server calls being blocking calls. For this run each flash sequence was 24000 pulse / 40 hz = 600s [10 mins]. The client timeout is 360s. 

On inspection of the code, the fact that the light was not stopped is due to the request to stop the deadman's switch ((which posts to the interlock server) being queued on that blocked thread due to some Objective c thread inheritance stuff (subtleties of when one thread is launched by another thread). 

I'll get this tidied up and put in a PR this weekend.

tlatorre-uchicago, 2017-10-31:
Fixed in #446.

closed 2017-10-31 by tlatorre-uchicago