snoplus/orca #457: Beach balling after positing a notification

EdLeming, 2017-10-25:
This issue is not so much an issue, but a note on a strange behaviour observed during development.

A mininum working example would be something like:

In SNOPController.m:

- (IBAction) emergencySmellieStopAction:(id)sender
{
   dispatch_sync(dispatch_get_main_queue(), ^{
        [[NSNotificationCenter defaultCenter] postNotificationName:ORSMELLIEEmergencyStop object:self];
    });
}

In ELLIEModel.m:

- (void) registerNotificationObservers
{
    NSNotificationCenter* notifyCenter = [NSNotificationCenter defaultCenter];    
    [notifyCenter addObserver : self
                     selector : @selector(killKeepAlive:)
                         name : ORSMELLIEEmergencyStop
                        object: nil];
}

It was observed that, on the GUI button click, ORCA would beach ball. Adding break points suggests the notification got posted, but it never made it inside the killKeepAlive method. If I comment out the notification post in emergencySmellieStopAction the button click is successful - but obviously does nothing. This does not appear to be a bug with the actioning of the button itself. 

In the end I worked around by calling the killKeepAlive method in the ELLIEModel directly in the emergencySmellieStopAction method. I never got to the bottom of it.

tlatorre-uchicago, 2017-10-26:
Just looking at this now, I suspect this is because you're dispatching to the main queue on the main thread. You should only ever use dispatch_sync when running from another thread. In this case if emergencySmellieStopAction is only called from the GUI it is guaranteed to be on the main thread.

EdLeming, 2017-10-26:
Yeah, I did have some thoughts along those lines. I tried removing the dispatch_sync, but the beach-balling persisted.

Although, now I think about it, there was a second dispatch_sync in the killKeepAlive function called by the observer. I guess the selector would have to be launched on the main thread to reproduce this problem situation, though. That sounds unlikely.

tlatorre-uchicago, 2017-10-31:
Do you have a small code sample to reproduce it without the dispatch_sync? Also, another thing to note is that posting a notification doesn't cause the function which receives the notification to run on a separate thread. The notifications are posted on the main thread, so if there was a chance of stalling the GUI, posting a notification will not help.