snoplus/orca #487: Could not load ECAL docs

tannerbk, 2018-02-06:
When trying to load ECALs today we got (for both crate 12 and 13):

18-02-06 00:22:23 XL3 crate 13 didn't receive all the ECAL documents.
Missing slots: 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 

We did this from operug.

tannerbk, 2018-02-06:
We need this functionality for high voltage running Thur evening. I'm assuming this is the result of http->https?

tlatorre-uchicago, 2018-02-06:
If it's due to https, you should be able to change the couchdb address to point to dbug.sp.snolab.ca on port 6984 (assuming the ECALs were loaded to the newer version of couchdb). If they were loaded to the older version of couchdb they should probably be re-uploaded to the newer version. @mastbaum can probably comment on where they were uploaded given the address and port used in penn daq.

mastbaum, 2018-02-07:
@tannerbk @tlatorre-uchicago -- see email.

tannerbk, 2018-02-08:
Still could not get this working on the teststand.

mastbaum, 2018-02-08:
Can you list the steps to reproduce and I'll give it try?

tannerbk, 2018-02-08:
Thanks Andy. 

In the crate view there is a "Fetch ECAL to GUI" (or something very similar). If you click that and wait 10 seconds you'll get an error in the orca logs like the ones in my first comment in this thread. 

I then played around with the debugdb database settings in the settings tab of the standard run control panel. Changing the settings to the ones suggest in the email did not seem to help.

mastbaum, 2018-02-14:
It appears this was due to DB settings -- please reopen if it happens again with the recommended setup.

tannerbk, 2018-06-15:
Still having an issue with loading ECAL documents at the teststand. Can you remind me what the database settings are supposed to be? I've tried several different sets of DB settings and I always get:
"XL3 crate 19 didn't receive all the ECAL documents". Do we have what the settings for the teststand should be documented somewhere?