snoplus/orca #488: Run type word unset itself

rcl11, 2018-02-09:
Started my shift with the bits for "physics" and "bubbles on" set in the runtype word. Part way through shift wanted to switch on "unusual activity" bit. Clicked the bit for unusual activity and let the run roll over (I seemed to remember doing this last summer and it would allow the bits to be picked up on the next run, which is what i wanted - possibly I should have hit restart run but I didn't think this was necessary). However on rollover ORCA lost 'physics' and 'bubblers on' from the runtype word and just left me with usual activity bit. It required us to hit 'refresh standard runs' and then restart the run to get the 'physics' bit back (although not 'bubblers on'). Screenshot attached  shows what ORCA looked like after the run rollover.

[image]

jcaravaca, 2018-02-12:
The detector state reports 'None' run type word, so it is not a display issue. I tried to reproduce it at the teststand with no luck. Let's keep an eye on this in case it happens again.
The changes in the run type word are only effective if the user clicks RESTART.

rcl11, 2018-02-12:
The afternoon shifter had an opportunity to try this as well (the unusual activity bit had been unset by mistake again, so he wanted to turn it back on) but the only difference was that in his case he clicked "reload standard runs" after setting the bit and in his case when the run rolled over it was picked up correctly and no other bits were lost. BTW I asked the 2 other shifters I saw and both were under the impression that the correct workflow was to let the run rollover rather than click restart run (maybe this used to be the case back in the summer? None of us had been on shift for a while).