snoplus/orca #505: cannot start runs without TUBii

tlatorre-uchicago, 2018-05-01:
ORCA should be able to start runs without TUBii, but on the teststand currently it seems as if you cannot start runs without TUBii. To reproduce, open the TUBii GUI and change the hostname to a non existent hostname and try to start a run. You should get the following error:

```
Run Not Started because of exception: RedisClient, reason: Operation timed out
```

I'm *pretty* sure that this used to work. @EdLeming any idea if any recent changes might have affected this?

EdLeming, 2018-05-01:
I don't believe there have been any recent changes that will affect this. Looking through the code there are two places in the SNOPModel which try to interact with the TUBii server l without an immediate try, catch (hence it's caught higher up the line in the ORRunModel - where that message originates) . Both calls are inside the runInitialization method :

https://github.com/snoplus/orca/blob/master/Source/Experiments/SNOP/SNOPModel.m#L761
(From [a commit last August](https://github.com/snoplus/orca/commit/3a5b00619bdb345d9b3d52915cd1b8333144acc3))

https://github.com/snoplus/orca/blob/master/Source/Experiments/SNOP/SNOPModel.m#L808
(From [a different commit last August](url))

Both commits are associated with adding TUBii info to standard runs. I imagine those are the culprits. I can test in the morning.

tlatorre-uchicago, 2018-05-01:
Ah good point. I'll have to go back and look to see when we ran without TUBii. Perhaps it was before TUBii was added to the standard runs.